adafruit/Adafruit_CircuitPython_LIS3DH #2: spinner_advanced memory allocation failed with Adafruit CircuitPython 1.0.0-rc.1

dglaude, 2017-06-20:
Using Adafruit CircuitPython 1.0.0-rc.1 and release 0.1.1 f9f906b
example spinner_advanced fail on memory allocation:

>>> import spinner_advanced
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "spinner_advanced.py", line 227, in <module>
MemoryError: memory allocation failed, allocating 376 bytes

Just trying to follow the https://www.youtube.com/watch?v=Qms1_QSdj6I

tannewt, 2017-06-21:
How are you loading the file? AMPY or through the USB drive?

dglaude, 2017-06-21:
Certainly not AMPY. Copy of file on USB from OSX.

jerryneedell, 2017-06-21:
I had the same problem if I tried to import the spinner_advanced example. The tutorial has it renamed to main.py to run as boot and that seems to be just enough memory difference to make it. I tried removing a bit of the code and am now able to run it via import.

dglaude, 2017-06-21:
Thanks, now working for me as main.py.
That memory limitation is really an annoyance.

tannewt, 2017-06-22:
Yup! Memory is tight on the device.

Another way to minimize memory use on import is to run it through mpy-cross. That will pack it into byte code which takes less memory to load.

closed 2017-06-22 by tannewt