adafruit/Adafruit_CircuitPython_LIS3DH #10: Reading acceleration data

pambrose, 2017-10-05:
Using [this](https://learn.adafruit.com/circuitpython-hardware-lis3dh-accelerometer/software) as a reference, I am trying to print accelerometer data from a Circuit Playground Express in a CircuitPython REPL. 

This script is printing values, but the values are not changing despite moving 
the Circuit Playground. Any feedback would be appreciated.

```python
import time

import board
import busio
import adafruit_lis3dh

i2c = busio.I2C(board.ACCELEROMETER_SCL, board.ACCELEROMETER_SDA)
lis3dh = adafruit_lis3dh.LIS3DH_I2C(i2c, address=25)
lis3dh.range = adafruit_lis3dh.RANGE_2_G

i = 0
while True:
    x, y, z = lis3dh.acceleration
    print('{} x = {}G, y = {}G, z = {}G'.format(i, x / 9.806, y / 9.806, z / 9.806))
    i = i + 1
    time.sleep(0.25)
```

mrmcwethy, 2017-10-05:
What version of CircuitPython and Adafruit Bundle are you using.  I am using CircuitPython and your example is working for me.

caternuson, 2017-12-06:
Try accessing the accelerometer through the Circuit Playground Express library:
https://learn.adafruit.com/circuit-playground-express-piano-in-the-key-of-lime/print-hello-circuitpython#installing-the-circuit-playground-express-library
and here's an example of printing the data:
https://learn.adafruit.com/how-tall-is-it/hello-accelerometer?view=all#using-circuitpython-repl

closed 2017-12-06 by caternuson