adafruit/Adafruit_CircuitPython_LIS3DH #12: Access to single axis value

caternuson, 2017-11-18:
This mainly comes up in conditionals, and can currently be done like this:
```python
if lis3dh.acceleration[2] > SOME_VALUE:
    print("Z axis exceeds the value!")
```
but requires knowing the index of the axis. Is there anyway this could be done where it's more obvious what's being accessed?
Like this:
```python
if lis3dh.acceleration['Z'] > SOME_VALUE:
    print("Z axis exceeds the value!")
```
or this:
```python
if lis3dh.acceleration.Z > SOME_VALUE:
    print("Z axis exceeds the value!")
```

dhalbert, 2017-11-18:
We could use `ucollections.namedtuple()`: http://circuitpython.readthedocs.io/en/2.x/docs/library/ucollections.html

Note that we are planning to rename all the `u...` modules eventually and make sure they match or are a subset of the CPython equivalents.

Sigafoos, 2018-01-07:
Would it be better to have `lisd3h.acceleration.z`?

tannewt, 2018-01-24:
Someone want to switch this to a named tuple and update the design guide?

Sigafoos, 2018-01-29:
I can!

PR #27 (2018-03-19): Adding a named tuple for acceleration
Fixes #12
closed 2018-03-19 by kattni