adafruit/Adafruit_CircuitPython_LIS3DH #16: Shake Function

platinum3x, 2017-12-14:
When calling the shake function: `lis3dh.shake()` the result always returns `False`. This happens even when vigorously shaking the lis3dh accelerometer device. The return of the shake function is `total_accel > shake_threshold`, so I'm assuming the values for `total_accel` are always GREATER THAN the `shake_threshold`. It would seem that I need to modify the parameter values of the shake function to increase the sensitivity of detecting a shake event. These parameters include `shake_threshold, avg_count, and total_delay`. However, I'm not sure how to write the appropriate code needed to change these parameters. Also, since the function is being called from a module in .MPY format, I am unable to edit the raw .MPY file to change the default parameters for this function. When using a .PY formatted module file instead of the .MPY module file, I receive a memory error message in the REPL (so I'm forced to work off of the .MPY file). Any help with figuring this out would be much appreciated!

Source code for the shake function within the lis3dh.mpy module file: [https://github.com/adafruit/Adafruit_CircuitPython_LIS3DH/releases/tag/v1.0.0](url)

Screenshot of shake function code within the module file:
[image]

caternuson, 2017-12-14:
> However, I'm not sure how to write the appropriate code needed to change these parameters. 

Here's the syntax to use in your calling code:
```python
lis3dh.shake(shake_threshold=YOUR_SHAKE_THRESHOLD, avg_count=YOUR_AVG_COUNT, total_delay=YOUR_TOTAL_DELAY)
```
Just replace the **YOUR_*** place holders with what values you want to use.

platinum3x, 2017-12-14:
Wow, thanks so much! I'm very new to all of this, but I got pretty close. Is there a way to call back the parameter values once they are set? For example, if I didn't know what `shake_threshold` was set to, what code would I use to identify that value?

caternuson, 2017-12-14:
There's currently no way to read the values back. The defaults are given in the documentation, so if you do nothing, that's what they are. To change them, you would have to provide the values, so then you would know what they are since you provided them. (at least that's the general idea)

caternuson, 2017-12-14:
Also, if you didn't want to change all 3 of the parameters, you can use this syntax when calling shake:
```python
lis3dh.shake(total_delay=YOUR_TOTAL_DELAY)
```
The parameters not specified will have default values.

closed 2017-12-14 by caternuson

platinum3x, 2017-12-14:
Thanks so much for the help! I spent too much time trying to figure this out haha. Much appreciated!

tannewt, 2017-12-14:
Thanks Carter! platinum3x I'm glad you got it figured out. We are usually
on Discord chat if you have more questions: https://adafru.it/discord or
the Adafruit support forums: https://forums.adafruit.com/viewforum.php?f=60