adafruit/Adafruit_CircuitPython_LIS3DH #21: return values in m/s^2 to match Arduino API

Sigafoos, 2018-01-07:
The Arduino LIS3DH library's `acceleration` is [returned in m/s^2](https://github.com/adafruit/Adafruit_LIS3DH/blob/master/examples/acceldemo/acceldemo.ino#L58). The CircuitPython library [converts it to Gs before returning](https://github.com/adafruit/Adafruit_CircuitPython_LIS3DH/blob/master/adafruit_lis3dh.py#L141)

Is there a reason for this? I'm not saying CP *must* match Arduino, but it seems like translating to Gs would be done by the user.

tannewt, 2018-01-10:
It is m / S^2 because its multiplying by 9.806. For clarity it really should be `(x / divider) * 9.806`

(and yes it should be m / s ^ 2 to match our [design guide](https://circuitpython.readthedocs.io/en/latest/docs/design_guide.html#sensor-properties-and-units) too.

Sigafoos, 2018-01-10:
Wow, I just utterly failed to understand what was going on. Yes, multiplying by 9.806 (which happens in *both* CPy and Arduino) makes it m/s^2, and the example *divides* it by 9.806 to convert back to Gs.

Thanks!

closed 2018-01-10 by Sigafoos