adafruit/Adafruit_CircuitPython_LIS3DH #22: add orientation property

Sigafoos, 2018-01-07:
The LIS3DH supports returning orientation data but the CP library does not expose this (nor, it seems, [does the Arduino library](https://github.com/adafruit/Adafruit_LIS3DH/blob/master/Adafruit_LIS3DH.cpp#L118))

This seems like something useful to include.

deshipu, 2018-01-07:
I can't see that register in the datasheet, maybe it's copied from some other model?

Sigafoos, 2018-01-08:
I noticed that later too. Also, I was right and my expectations were wrong: as you said in Discord, you only get the acceleration readings and can try to determine orientation from that.

Lawmate, 2019-06-07:
Hi, I'm trying to get the orientation function of the LIS3DH working. There's a register called INT1_CFG where you should be able to set it up, then read it from INT1_SRC, but I'm having no success. It's always reutrning 0. The function getOrientation is listed in the header file of the arduino library but there's no corresponding function in the .cpp. Do you guys know if this is because no one could get it working either? Thanks

caternuson, 2019-06-07:
^^ also - https://forums.adafruit.com/viewtopic.php?f=19&t=152690

caternuson, 2019-06-07:
I'm re-opening this for discussion.

@Lawmate what benefit would that have vs. just reading acceleration and doing some orientation logic in your code?

Lawmate, 2019-06-07:
Hey, sorry for asking in 2 locations.

It definitely could be done in the micro code, but it's a capability that's offered by the chip and is advertised by Adafruit.

It would take a processing burden off the micro by having having an orientation change being communicated by an interrupt driven event. Plus it's an algorithm that's already been implemented within the chip, so takes care of that development. It's not the most complicated thing to sort on the micro but seeing as it's a feature that is offered by ST on the chip, it seems a bit tautological to have to reimplement it on the micro. Am happy to go the easiest and most direct route though, not wedded to making it work.

caternuson, 2019-06-10:
OK, makes sense. So now let's talk implementation details. The general request was for an `orientation` property. Focusing on just that for now, what exactly would that return?

Lawmate, 2019-06-12:
In the LIS3DH datasheet, there's a register INT1_CFG (0x30) that as I understand it, can be used to configure what the interrupt 1 pin does. Once that has been set up, and the interrupt is triggered, there is the register INT1_SRC (0x31) that can be read. The last 6 bits of this register are whether X, Y or Z is high or low. As I understand it, and depending on the exact configuration, the interrupt is triggered when the device is oriented with one of these axes pointing vertically. Or once movement has finished, the axis that is closest to vertical will be a logic 1 in that register.

So I'm guessing the library would initilise the interrupt, then set up an ISR to update a variable?

Not 100% sure of the details of this, but either way, I can't get the interrupt to trigger. Have tried various configurations.

Thanks for the help.

caternuson, 2019-06-12:
CircuitPython does not support interrupts. So there's no way to have an ISR run. In general, some form of polling approach must be taken. So you'd get the state of the interrupts (the `INT1_SRC` register) via something like this:
```python
src = lis3dh.interrupt_source
```
which would return....?? Could be a tuple of booleans for each bit in the register. Or a dictionary with the keys being the bit entry names (like `ZA`). Or some other approach. That's the general discussion.

Would polling work for your application? Or do you require true interrupt and ISR functionality?

Lawmate, 2019-06-13:
Aha, ok no problemo. Really appreciate the response. I'll ask this on the arduino library repo.

caternuson, 2019-06-13:
It sounds like polling would not work for what you are trying to do?

geekguy-wy, 2020-02-06:
Usually, orientation is considered to consist of pitch, roll, and heading. All are usually in degrees. It does not _look_ like this is what the chip is returning. Personally, I prefer the calculations to be done on the micro or these reading be returned from sensor fusion on a chip because the data returned is much more useful. This is more part of a full AHRS system though.

kattni, 2020-04-30:
Closing this.

closed 2020-04-30 by kattni