adafruit/Adafruit_CircuitPython_LIS3DH #28: tap.py example not working

jerryneedell, 2018-03-19:
I tried the tap.py example on CP2.2.4 and CP3.0-alpha on a CPX with the latest and previous version of this driver and I cannot get it report any taps.
Not much info to privide - the script executes but never responds to taps (even no gentle ones).

Does it work for anyone else?

Sigafoos, 2018-03-19:
Did you modify the file at all? By default it uses `board.SCL` and `board.SDA` which are the external I2C pins. On a CPX you need to uncomment lines 16-19.

I just tried that, though, and I also am not getting any output. Let's look at `tapped` more in depth...

jerryneedell, 2018-03-19:
Yes, I set it up for CPX. Thanks for digging in!

Sigafoos, 2018-03-19:
I think the issue is that there isn't enough of a delay in the loop, so the code never gets checked. I added a half second delay and the requirement that the last check not be `True`.

I'm not saying this is the only way to solve it, but it does work.

jerryneedell, 2018-03-20:
Did something change to make this stop working.  That is, why is the delay necessary now if it was not necessary in the past?

jerryneedell, 2018-03-20:
I see that #28 made the real change to the code.  I have not had a chance to try it - will try it this evening.

jerryneedell, 2018-03-20:
I'm a little confused by the proposed changes. The real fix is to removed the "double" tap test from the  example code and let the driver detect the double tap, correct?  Was the delay was just added for petter performance? I'll try to test this this evening.

Sigafoos, 2018-03-20:
I think I added the delay first, and then removed the double tap.

The "double" check was added not because the driver couldn't detect double taps but because it will report whether it's detected a tap as true for a certain period after the tap. I think that a delay does roughly the same thing.

I'm not necessarily advocating for this: it's just a crack I took at the issue.

caternuson, 2018-04-28:
Hopefully fixed by #32

closed 2018-04-28 by caternuson