adafruit/Adafruit_CircuitPython_LIS3DH #43: Missing adafruit-circuitpython-busdevice dependency

oexal, 2018-08-29:
This package requires the package adafruit-circuitpython-busdevice but the package is not configured to requires this package. It result on this error when we try to run the package on a clean raspbian install.

pi@raspberrypi:~ $ sudo python3 ./accel.py
Traceback (most recent call last):
  File "./accel.py", line 9, in <module>
    lis3dh = adafruit_lis3dh.LIS3DH_SPI(spi, cs)
  File "/usr/local/lib/python3.5/dist-packages/adafruit_lis3dh.py", line 347, in __init__
    import adafruit_bus_device.spi_device as spi_device
ImportError: No module named 'adafruit_bus_device'

---------------------------------------------------------------------------------------------

pi@raspberrypi:~ $ pip3 show adafruit-circuitpython-lis3dh
Name: adafruit-circuitpython-lis3dh
Version: 4.3.1
Summary: CircuitPython library for LIS3DH accelerometer.
Home-page: https://github.com/adafruit/Adafruit_CircuitPython_LIS3DH
Author: Adafruit Industries
Author-email: circuitpython@adafruit.com
License: MIT
Location: /usr/local/lib/python3.5/dist-packages
Requires: Adafruit-Blinka

this last line should show 
Requires: Adafruit-Blinka, adafruit-circuitpython-busdevice

Thank you!

ladyada, 2018-08-29:
good catch!

ladyada, 2018-08-29:
fixed and re-released!

closed 2018-08-29 by ladyada