adafruit/Adafruit_CircuitPython_LIS3DH #52: lis3dh_simpletest does not work for pygamer

jerryneedell, 2019-07-01:
The current lis3dh_simpletest example uses  the existence of ACCELEROMETER_SCL to determine the pin configuration and I2C address
https://github.com/adafruit/Adafruit_CircuitPython_LIS3DH/blob/master/examples/lis3dh_simpletest.py#L9

```
if hasattr(board, 'ACCELEROMETER_SCL'):
    i2c = busio.I2C(board.ACCELEROMETER_SCL, board.ACCELEROMETER_SDA)
    int1 = digitalio.DigitalInOut(board.ACCELEROMETER_INTERRUPT)
    lis3dh = adafruit_lis3dh.LIS3DH_I2C(i2c, address=0x19, int1=int1)
else:
    i2c = busio.I2C(board.SCL, board.SDA)
    int1 = digitalio.DigitalInOut(board.D6)  # Set to correct pin for interrupt!
    lis3dh = adafruit_lis3dh.LIS3DH_I2C(i2c, int1=int1)
```

But this does not work for the pygamer. It has ACCELEROMETER_INTERRUPT but uses the default board SCL/SDA.

I'll be happy to put in a PR but wanted to see how you wanted to implement it or should there just be a separate lis3dh_pygamer_simpletest?

jerryneedell, 2019-07-01:
this example works for the pygamer:
```
import time
import board
import digitalio
import adafruit_lis3dh

int1 = digitalio.DigitalInOut(board.ACCELEROMETER_INTERRUPT)
lis3dh = adafruit_lis3dh.LIS3DH_I2C(board.I2C(), address=0x19, int1=int1)

# Set range of accelerometer (can be RANGE_2_G, RANGE_4_G, RANGE_8_G or RANGE_16_G).
lis3dh.range = adafruit_lis3dh.RANGE_2_G

# Loop forever printing accelerometer values
while True:
    # Read accelerometer values (in m / s ^ 2).  Returns a 3-tuple of x, y,
    # z axis values.  Divide them by 9.806 to convert to Gs.
    x, y, z = [value / adafruit_lis3dh.STANDARD_GRAVITY for value in lis3dh.acceleration]
    print('x = {}G, y = {}G, z = {}G'.format(x, y, z))
    # Small delay to keep things responsive but give time for interrupt processing.
    time.sleep(0.1)
```

kattni, 2019-07-01:
Put in a PR. I'd rather not have a separate `simpletest` file for PyGamer. Two options I can think of:
* Use the lack of `ACCELEROMETER_SDA` and the presence `ACCELEROMETER_INTERRUPT` to  modify the current `if` statement, and add another statement that identifies PyGamer.
* Add another setup section commented out with a comment identifying it as the PyGamer setup (we've done this many times, the current setup in LIS3DH is unusual, specific to CPX essentially)

I'll let you decide how you'd like to do it, but a PR would certainly be appreciated!

caternuson, 2019-07-01:
Why does the one case also have an explicit address set?

How about moving the `int1` setup out of this `if:else` and into a separate one above it? Also, switch to using the I2C singleton for default case.

```python
# configure interrupt pin
if hasattr(board, 'ACCELEROMETER_INTERRUPT'):
    int1 = digitalio.DigitalInOut(board.ACCELEROMETER_INTERRUPT)
else:
    int1 = digitalio.DigitalInOut(board.D6)  # Set to correct pin for interrupt!

# configure I2C bus
if hasattr(board, 'ACCELEROMETER_SCL'):
    i2c = busio.I2C(board.ACCELEROMETER_SCL, board.ACCELEROMETER_SDA)
else:
    i2c = busio.I2C()

# create LIS3D object
lis3dh = adafruit_lis3dh.LIS3DH_I2C(i2c, address=0x19, int1=int1)
```

jerryneedell, 2019-07-01:
The default address is 0x18

caternuson, 2019-07-01:
So the logic is trying to actually infer the specific board - CircuitPlaygroud, PyGamer, etc., so it can then also set the address as needed? This could become a fairly nasty logic blob to cover all possible combinations of I2C pin names, interrupt pin names, and addresses that might be used.

For the sake of keeping this "simple", could just cover the majority case and have code comments that say "change pins if needed" and "change address if needed". So for PyGamer, user would need to edit code.

jerryneedell, 2019-07-01:
Agreed! I’ll put in a PR this evening. I just wanted to document the difference

PR #53 (2019-07-03): add PyGamer configuration to lis3dh_simpletest.py
Add commented lines to example for PyGamer I2C configuration
Fixes #52
closed 2019-07-03 by ladyada