adafruit/Adafruit_CircuitPython_LIS3DH #67: Small INT overflow on Halowing_m0

jerryneedell, 2021-02-01:
With current tip of main CP installed on a Hallowing_m0  __init__ fails:
```
Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 6.2.0-beta.1-23-g0c0b51711 on 2021-02-01; HalloWing M0 Express with samd21g18
>>> import lis3dh_simpletest
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "lis3dh_simpletest.py", line 16, in <module>
  File "adafruit_lis3dh.py", line 332, in __init__
OverflowError: small int overflow
>>> 
```

ladyada, 2021-02-01:
@jerryneedell kinda weird, that line is the i2c initialor??

jerryneedell, 2021-02-01:
> @jerryneedell kinda weird, that line is the i2c initialor??

Agreed!

ladyada, 2021-02-01:
wanna try tracking down?

jerryneedell, 2021-02-01:
sure -- just pulling out my breakout and some other boards to examine.

jerryneedell, 2021-02-01:
Works OK on a qtpy_m0_haxpress...

jerryneedell, 2021-02-01:
hmmm -- maybe related to this https://github.com/adafruit/circuitpython/blame/main/ports/atmel-samd/boards/hallowing_m0_express/mpconfigboard.mk#L12 ...  added about 6 months ago though -- I'll try some tests

jerryneedell, 2021-02-01:
Yes -- changing 
LONGINT_IMPL = NONE
to
LONGINT_IMPL = MPZ

fixes it, BUT it won't fit in flash -- I had to remove pulseio to test it.

Now to find out why this makes a difference. Is I2C broken on any board that has 
LONGINT_IMPL = NONE
That is a lot of boards....

jerryneedell, 2021-02-01:
ruh roh...
It looks like an issue with the native busdevice
I build with CIRCUITPY_BUSDEVICE = 0
an use the Adafruit_BusDevice Library it works OK
... I'll file an issue in the CP repo

ladyada, 2021-02-01:
oi good find!@

tannewt, 2021-02-02:
Closing in favor of https://github.com/adafruit/circuitpython/issues/4109

closed 2021-02-02 by tannewt